chester4444/esp-multical21 #1: #include "credentials.h"

dkcsn, 2021-02-12:
This seems really interesting - 
Would it be possible to share the template of credentials.h?

chester4444, 2021-02-12:
Replace your Wifi setting(s), your meter AESkey and the meter serial number

```
// defines for the credentials
#define SSID1   "ssid1"
#define PW1     "pass1"
#define MQTT1   "10.0.0.100"
#define SSID2   "ssid2"
#define PW2     "pass2"
#define MQTT2   "10.0.0.100"
#define SSID3   "IoT"
#define PW3     "pass3"
#define MQTT3   "10.14.0.1"
#define W_ENCRYPTION_KEY      0x00, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff
#define W_SERIAL_NUMBER       0x12, 0x34, 0x056, 0x78
```

dkcsn, 2021-02-12:
Thank you :-)

This Crypto.h -
i tried https://rweather.github.io/arduinolibs/crypto.html
it this the right one ? - i'm getting 
/Users/christian_nielsen/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/Crypto.h:749:5: error: 'br_hkdf_context' in namespace '::' does not name a type
     ::br_hkdf_context hkdfContext;

dkcsn, 2021-02-12:
I found out - the code is compatible for ESP8266 up to version 2.6.3 - version 2.7.0 and up fails compiling.
so you need to choose maximum 2.6.3 in  https://arduino.esp8266.com/stable/package_esp8266com_index.json
with that said - amazing work

dkcsn, 2021-02-12:
__

> Replace your Wifi setting(s), your meter AESkey and the meter serial number
>
> ```
> // defines for the credentials
> #define SSID1   "ssid1"
> #define PW1     "pass1"
> #define MQTT1   "10.0.0.100"
> #define [...]

Thank you for providing this 🤟 - wondering why you can/shall define wifi/PW/MQTT three times ?

chester4444, 2021-02-12:
Im using PlatformIO und i am still at 2.6.3 - thanks for the hint.
3 Wifis? it was easier with them in the development phase.

dkcsn, 2021-02-27:
Today I recieved my KEM files to the Kamstrup  Multical 21, it took a little time to find out how to decrypt and get the DEK key (W_ENCRYPTION_KEY) exported.
Still waiting for the CC1101 module :-) I should arrive shortly.  Wondering if I can use a Username and Password for the MQTT server - i looks like U/P is for wifi only?

chester4444, 2021-02-28:
to connect to the mqtt-server with user/password, change
   https://github.com/chester4444/esp-multical21/blob/f6841f9f4770733a32cedc598a8f902f6fdbc16d/src/main.cpp#L187
to
`  return mqttClient.connect(ESP_NAME, "mqtt-user", "mqtt-password", "/watermeter/online", 0, true, "False");`

dkcsn, 2021-04-14:
Thank. you .- this works great :-)

closed 2021-04-14 by dkcsn

atifan, 2021-07-08:
Hello. How can i set the port for mqtt-server?

atifan, 2021-07-09:
@dkcsn : Can u tell my how u did export the DEK ? I am waiting for my key from waterworks.
Where did you get the serial number? From the device? In my case the S/N looks like 12345678/TR/20 -> how did you convert that in hex or what, for the credentials.ini ?