chester4444/esp-multical21 #2: ESP32

CentauriDK, 2021-03-13:
Will their also come a version for the ESP32 ?
I have tryed for the WeMos D1 MINI ESP32 but just get a bounche of errors.

chester4444, 2021-03-14:
at least now it compiles for ESP32, but i cant test it (in near future).
edit _**hwconfig.h**_ and double check pin connections

CentauriDK, 2021-03-18:
Thanks
I had to change the platformio,ini to be able to compile 
from:
lib_deps = rweather/Crypto @ ^0.2.0	
to:
lib_deps = 
	rweather/Crypto @ ^0.2.0
	knolleary/PubSubClient@^2.8
I just get an couple of warnings now when compiling. But it can compile.

And maybe you should include the credentials.h file and add some dummy data, it can confuse many that they have to create it by them self.

CentauriDK, 2021-03-28:
I have now testet it on my WeMos D1 MINI ESP32
I get no communication on the serial terminal at all. So can only assume their is som sort of cinflict.
It works fine on the WeMos D1 MINI ESP8266

atifan, 2021-07-07:
Hello. I want to use this firmware too, but I am missing an example file for credentials.h @CentauriDK could you post an example for it here please? Or write me an E-Mail -> mrosiris@gmx.de. Would be very nice, I canot compile because I dont know how the credentials.h must have in.

chester4444, 2023-11-28:
New version now features a config-template.h
Fill in the gaps and rename it to config.h