chester4444/esp-multical21 #5: Picture of ready hardware?

atifan, 2021-07-06:
Hello, This device is exactly what I need. I want to read Multical21 and send the data to MQTT-Broker.
What I miss is a picture of ready built hardware. What pins do I need to connect?
And a description what I have to change in the source code before comping.
Would be nice if someone could explan and make a picture or video :)
Thanks.

pazzotranquillo, 2022-09-06:
> Hello, This device is exactly what I need. I want to read Multical21 and send the data to MQTT-Broker. What I miss is a picture of ready built hardware. What pins do I need to connect? And a [...]

Hi, have you sorted this out? I'd like to build it but I have no clue how to connect the CC1101 to my ESP8266.

chester4444, 2022-09-07:
this should help:
https://github.com/chester4444/esp-multical21/blob/b076743f544bb2a55108afe9febdb6e9f65013bd/include/hwconfig.h#L5

closed 2023-04-02 by chester4444