chester4444/esp-multical21 #7: no WLAN-connection

atifan, 2021-07-09:
I can compile, and upload to a WEMOS D1 Mini, but the WEMOS does not connect ty my WLAN. In credentials.h there is set the SSID1 and the Password.
Any ideas why it is not working?

CentauriDK, 2021-07-09:
What does the Serial Console say ?
It will normally output some debug information on scanning and connecting to the wifi.

atifan, 2021-07-09:
Hi it says

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffda0 end: 3fffffc0 offset: 01a0
3fffff40:  40208ef5 000000f5 3ffee82c 4020130c  
3fffff50:  3fffdad0 000000f5 3ffee900 402013e9  
3fffff60:  000000ff 00000000 00000036 402013b7  
3fffff70:  3fffdad0 00000000 3ffee82c 40201471  
3fffff80:  0001c200 00000000 3ffee82c 40201660  
3fffff90:  3fffdad0 00000000 3ffee930 40201e58  
3fffffa0:  feefeffe feefeffe 3ffeea1c 402084f0  
3fffffb0:  feefeffe feefeffe 3ffe8528 401010b1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffda0 end: 3fffffc0 offset: 01a0
3fffff40:  40208ef5 000000f5 3ffee82c 4020130c  
3fffff50:  3fffdad0 000000f5 3ffee900 402013e9  
3fffff60:  000000ff 00000000 00000036 402013b7  
3fffff70:  3fffdad0 00000000 3ffee82c 40201471  
3fffff80:  0001c200 00000000 3ffee82c 40201660  
3fffff90:  3fffdad0 00000000 3ffee930 40201e58  
3fffffa0:  feefeffe feefeffe 3ffeea1c 402084f0  
3fffffb0:  feefeffe feefeffe 3ffe8528 401010b1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffda0 end: 3fffffc0 offset: 01a0
3fffff40:  40208ef5 000000f5 3ffee82c 4020130c  
3fffff50:  3fffdad0 000000f5 3ffee900 402013e9  
3fffff60:  000000ff 00000000 00000036 402013b7  
3fffff70:  3fffdad0 00000000 3ffee82c 40201471  
3fffff80:  0001c200 00000000 3ffee82c 40201660  
3fffff90:  3fffdad0 00000000 3ffee930 40201e58  
3fffffa0:  feefeffe feefeffe 3ffeea1c 402084f0  
3fffffb0:  feefeffe feefeffe 3ffe8528 401010b1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

atifan, 2021-07-09:
Hello, i fixed it.
Problem was that I did not connect the CC1101. It has to be connected to the ESP, that the ESP starts. When not connected, the ESP does not boot up.

closed 2023-04-02 by chester4444