chester4444/esp-multical21 #12: Compiled fine

dkcsn, 2023-12-07:
Hi Chester4444
Thank you for developing this great Integration to Kamstrup Multical21.
Looking forward to implement you updated, special the CRC check is great as the old implementation were a little unstabile (at least at my side)
Two Questions... 
Are you thinking af implementing Wifi Manager - and perhaps if the data can be read direct at the ESP at http://ip/data.json ? That would ease my Integration to my Fibaro Platform.
Second - you already wrote that CC1101 module for 868 MHz is needed.
I did my integration with the green module - works fine
The Blue T1 cc1101 with SMA antenna .- did you test that ? - my reason for asking is all the links and descriptions of that says 433Mhz. Also for the link in you answer in the first Issues request.
So I was wondering, did you alter it for 868 MHz or does it work even if everything on it says 433?

chester4444, 2023-12-07:
Hi dkcsn,

thanks for your feedback.

About the cc1101:
the blue module is basically working, nothing altered - but the green one does have a way better range. 

A webserver serving JSON should not be a big deal - the wifi manager is a little bit an overkill.
I'll see what I can do.

jnissen, 2023-12-11:
Hello, 
On compling an error occured. Can Not find wifi.sethostname in Main.Cpp.

if i comment this Out It Compiled Fine.