chester4444/esp-multical21 #15: Confirm for my personal setup and pcb :)

jascdk, 2024-01-10:
@chester4444 Hi Chester 

Great work on this project - happy to see the recent updates you made;)

I did - some while ago - make an pcb (just for trying it out:)). I made it compatible with the D1 mini or other same sized d1 mini clones and so on - and together with an CC1101 (433 mhz) version - the blue pcb with an sma-antenna.

I never came further with trying the electronics out - also because i read elsewhere that the 433 mhz board did not work. 

Together with your fresh updated code - it works;) I get the data i need - although the temp readings are off - i dont think it has been configured for my meter.

I post a picture of my board for your interest. I think about making a 3d printed case soon.

BTW - do you know if this project is possible to adapt over to esp-home? As i can see there is no offcial CC1101 integration yet. There is costum code for it here https://github.com/dbuezas/esphome-cc1101 , but i dont know if it can be coded to work with this project? 

And do you have a setup yaml code for Home Assistant? 

All the best

Jacob / Denmark

[image]

chester4444, 2024-01-10:
Hi Jacob, thanks for the feedback and congrats to your PCB, nice job. 
Sorry, I can't help with esp-home or Home Assistant - no experience.

I built my smarthome around smarthomeNG, homebridge, MQTT, ... on a raspberry.
Good luck

exetico, 2025-01-31:
@jascdk Hi! Did you simply stick to this, of have you migrated to a ESPHome based solution?

kakmoster, 2025-09-10:
> [@jascdk](https://github.com/jascdk) Hi! Did you simply stick to this, or have you migrated to a ESPHome based solution?
> 
> Also, while having you: How often do you pick up the "total" value? By monitoring the output, I've noticed it's not printing values that often.
> 
> Update: Now, after I've setup MQTT; it's clear how often the value is transmitted: [image]
> 
> Edit: I've moved to https://github.com/SzczepanLeon/esphome-components
> 
> Here's some sample code for that, with the most important settings.
> 
> ...
> esp32:
>   board: esp32dev
>   framework:
>     type: arduino
> 
> time:
>   - platform: sntp
>     id: time_sntp
> 
> external_components:
>   - source: github://SzczepanLeon/esphome-components@4.1.0 # v4+ requires ESP32
>     refresh: 0d
>     components: [ wmbus ]
> ....
> 
> wmbus:
>   all_drivers: True
>   log_all: True
>   mosi_pin: 23
>   miso_pin: 19
>   clk_pin: 18 # Serial ClocK (SCK)
>   cs_pin: 4 # CSN
>   gdo0_pin: 32
>   gdo2_pin: 33
> 
> sensor:
>   - platform: wmbus
>     meter_id: 0xSERIALNO
>     type: multical21
>     key: "REMOVED"
> 
>     sensors:
>       - name: "Cold Water RSSi"
>         field: "rssi"
>         accuracy_decimals: 0
>         unit_of_measurement: "dBm"
>         device_class: "signal_strength"
>         state_class: "measurement"
>         entity_category: "diagnostic"
> 
>       - name: "Cold Water Total"
>         field: "total"
>         accuracy_decimals: 3
>         unit_of_measurement: "m³"
>         device_class: "water"
>         state_class: "total_increasing"
>         icon: "mdi:water"
> 
>       - name: "Flow Temperature"
>         field: "flow_temperature"
>         accuracy_decimals: 1
>         unit_of_measurement: "C"
>         device_class: "temperature"
>         state_class: "total_increasing"
>         icon: "mdi:temperature"
> 
>       - name: "External Temperature"
>         field: "external_temperature"
>         accuracy_decimals: 1
>         unit_of_measurement: "C"
>         device_class: "temperature"
>         state_class: "total_increasing"
>         icon: "mdi:temperature"
> 
>       - name: "Max Flow"
>         field: "max_flow"
>         accuracy_decimals: 2
>         unit_of_measurement: "m3h"
>         device_class: "volume_flow_rate"
>         state_class: "total_increasing"
>         icon: "mdi:waves"

Could you post that entire esphome configuration? I'm getting a bunch of errors when compiling. Tried v5 and v4.