chester4444/esp-multical21 #21: Weird temperature values

sippe2, 2024-04-02:
I don't know is it just me...

I get really strange temperature values when reading. Water flow is exactly correct but temperature values are not correct:

Cold water (Multical21)
total: 415.257 m - target: 414.898 m -  0 C -  0 C - 0x00 
total: 415.257 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.257 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.257 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.257 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.259 m - target: 414.898 m -  0 C -  0 C - 0x00 
total: 415.259 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.259 m - target: 414.898 m - 84 C -  6 C - 0x00 
total: 415.259 m - target: 414.898 m - 84 C -  6 C - 0x00 

Hot water (Multical21)
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m -  0 C -  0 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00 
total: 270.839 m - target: 270.535 m - 32 C -  4 C - 0x00

alx6ths, 2024-10-18:
Hello, the same thing is happening to me, currently the water temperature varies between 14 and 97°C and the ambient temperature alternates between these values: 0, 14 and 85°C.

Maxxxe, 2025-05-22:
Here the same problem!
In there a solution?